balena-os/balena-intel #73: GIgabyte Brix

xanview, 2017-07-19:
I'm trying to boot ResinOS on the Gigabyte Brix GB-BXBT-1900 (https://www.gigabyte.com/Mini-PcBarebone/GB-BXBT-1900-rev-10) but for some reason I'm just getting a blank screen.

It is a bit strange, when I insert the hard drive into a NUC, it boots, when I then insert it into a Brix, it shows a blank screen, when I put the hard drive back into the NUC it just shows the Grub command line and stops booting. Not quite sure what is happening?

The Brix is quite similar to the NUC in terms of dimensions and specs and I am able to use Ubuntu Server on it. When using Ubuntu Server I normally put the following in  /etc/default/grub for both the NUC and Brix, otherwise, it sometimes fails to reboot:

```
GRUB_RECORDFAIL_TIMEOUT=30
GRUB_TIMEOUT=0
GRUB_CMDLINE_LINUX="text nomodeset intel_idle.max_cstate=1"
GRUB_TERMINAL="console"
```

Any advice how to troubleshoot?

xanview, 2017-07-20:
Ah, ignore that, I was flashing direct to the hard drive. When flashing from USB stick, it boots and installs as expected! - After this, I had to go into the bios and select UEFI boot mode and it booted, awesome :D

I'll report if there are any issues using the Brix instead of the NUC.

closed 2017-07-20 by xanview

agherzan, 2017-07-20:
Glad to hear this works nicely on that board too. I'm. Wondering if we could document this somewhere @shaunmulligan.

xanview, 2017-07-20:
The setup is identical to the NUC, been trying to reboot and also yanking the power cord out a few times, reboots like a champ :)

xanview, 2017-07-20:
I also tried the same on a Shuttle XH61V with a Pentium G2020 CPU, same process and results - all works beautifully :) -- it has 2 ethernet ports, and ResinOS gets on the network without issues:

```
root@resin:~# ifconfig -a
enp3s0    Link encap:Ethernet  HWaddr 80:EE:73:5A:86:01  
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

enp5s0    Link encap:Ethernet  HWaddr 80:EE:73:5A:86:00  
          inet addr:192.168.1.91  Bcast:192.168.1.255  Mask:255.255.255.0
          inet6 addr: fe80::1059:79c1:109c:7f19%119/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:4959 errors:0 dropped:0 overruns:0 frame:0
          TX packets:107 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:313076 (305.7 KiB)  TX bytes:20202 (19.7 KiB)
```

agherzan, 2017-07-20:
Thanks for the feedback. Enjoy!