balena-os/balena-intel #74: NUC DN2820FYK Grub Prompt on Reboot

xanview, 2017-07-19:
Hi there,

I followed these steps trying to use the 2.0.6-rev1 version for NUC:

```
$ time sudo resin local flash resin.img --drive /dev/disk4
? This will erase the selected drive. Are you sure? Yes
Flashing [========================] 100% eta 0s 
Validating [========================] 100% eta 0s 
sudo resin local flash resin-nuc-2.0.6-rev1.img --drive /dev/disk4  59.61s user 24.87s system 77% cpu 1:48.35 total
```

I inserted the hard drive into my NUC DN2820FYK (the older one from 2015 using 12v power). It seems to boot OK, but when I ssh it shows this:

```
local ➜  ~  ssh root@resin.local -p22222
ssh_exchange_identification: read: Connection reset by peer
local ➜  ~  
local ➜  ~  ssh root@resin.local -p22222
ssh: connect to host resin.local port 22222: Connection refused
```

So it seems on first connect it fails to ssh and then the ssh process dies completely? -- I also tried connecting an HDMI monitor and alt+f1 just shows the Resin splash screen (I tried waiting a while) and alt+f2 etc shows a blank screen.

When I try to reboot the unit, it can no longer boot and just shows me the Grub prompt :(

Any ideas?

xanview, 2017-07-19:
Oh, I misunderstood, I flashed this direct onto a hard drive instead of a USB storage, closing. Is there an image that I can flash direct onto the hard drive?

closed 2017-07-19 by xanview

agherzan, 2017-07-19:
@xanview indeed. We only provide flasher images for NUC at this moment. We are working though in offering both images in the near future.