balena-os/balena-intel #107: Include firmware for Intel® Wireless-AC 3168 M.2

floion, 2018-02-16:


harborretail, 2020-01-07:
Hey @floion can you also include Intel® Wireless-AC 3168 for the Intel image(s)?

Or can you direct me on how to get that firmware functioning?

We're using this intel-image (NUC) on the 2GB/32GB version of the LattePanda SBC. Generally it's been highly compatible but getting the wifi chip working (3165D2W) would be extremely helpful.

janvda, 2020-01-14:
Hi,

I am facing the same problem (see  https://forums.balena.io/t/intel-nuc-kit-nuc6cayh-intel-celeron-j3455-wifi-not-provisioning/2585/31?u=janvda) 
I have copy pasted the contents of my forum post below : 

Wifi is also not working for me on my **Intel NUC Kit NUC6CAYS**
I am using `balenaOS 2.46.0+rev1`

`dmesg` shows the following output.

```
[    3.979067] Intel(R) Wireless WiFi driver for Linux
[    3.979070] Copyright(c) 2003- 2015 Intel Corporation
[    3.979183] iwlwifi 0000:02:00.0: enabling device (0000 -> 0002)
[    3.980485] [drm] Cannot find any crtc or sizes
[    3.980571] iwlwifi 0000:02:00.0: Direct firmware load for iwlwifi-3168-29.ucode failed with error -2
[    3.980600] iwlwifi 0000:02:00.0: Direct firmware load for iwlwifi-3168-28.ucode failed with error -2
[    3.980629] iwlwifi 0000:02:00.0: Direct firmware load for iwlwifi-3168-27.ucode failed with error -2
[    3.980652] iwlwifi 0000:02:00.0: Direct firmware load for iwlwifi-3168-26.ucode failed with error -2
[    3.980676] iwlwifi 0000:02:00.0: Direct firmware load for iwlwifi-3168-25.ucode failed with error -2
[    3.980703] iwlwifi 0000:02:00.0: Direct firmware load for iwlwifi-3168-24.ucode failed with error -2
[    3.980733] iwlwifi 0000:02:00.0: Direct firmware load for iwlwifi-3168-23.ucode failed with error -2
[    3.980757] iwlwifi 0000:02:00.0: Direct firmware load for iwlwifi-3168-22.ucode failed with error -2
[    3.980760] iwlwifi 0000:02:00.0: no suitable firmware found!
[    3.980762] iwlwifi 0000:02:00.0: minimum version required: iwlwifi-3168-22
[    3.980764] iwlwifi 0000:02:00.0: maximum version supported: iwlwifi-3168-29
[    3.980765] iwlwifi 0000:02:00.0: check git://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git
```

Any help would be much appreciated.
Jan

floion, 2020-01-31:
Done in https://github.com/balena-os/balena-intel/pull/276. Will be available in 2.47.0+rev1 or newer

closed 2020-01-31 by floion