balena-os/balena-intel #123: Include firmware for Realtek RTL8723BS Wireless?

robp2175, 2018-04-15:
I am in the process of trying to create the build myself, but it would be much appreciated if you folks would consider adding this firmware. Thank You
https://github.com/hadess/rtl8723bs

robp2175, 2018-04-21:
It actually appears is though this firmware is already there and when I run dmesg there does not appear to be any mention of this hardware. lspci also shows nothing. Please let me know what other information you need.

```
root@c2b874e:/usr/src/app# ls -la /lib/firmware/rtlwifi
total 489
drwxr-xr-x 2 root root  1024 Mar 29 11:13 .
drwxr-xr-x 9 root root  3072 Mar 29 11:13 ..
-rw-r--r-- 1 root root 13904 Mar 15 09:36 rtl8188eufw.bin
-rw-r--r-- 1 root root 16014 Mar 15 09:36 rtl8192cufw.bin
-rw-r--r-- 1 root root 16116 Mar 15 09:36 rtl8192cufw_A.bin
-rw-r--r-- 1 root root 16096 Mar 15 09:36 rtl8192cufw_B.bin
-rw-r--r-- 1 root root 16116 Mar 15 09:36 rtl8192cufw_TMSC.bin
-rw-r--r-- 1 root root 22172 Mar 15 09:36 rtl8723aufw_A.bin
-rw-r--r-- 1 root root 24118 Mar 15 09:36 rtl8723aufw_B.bin
-rw-r--r-- 1 root root 19200 Mar 15 09:36 rtl8723aufw_B_NoBT.bin
-rw-r--r-- 1 root root 30746 Mar 15 09:36 rtl8723befw.bin
-rw-r--r-- 1 root root 31762 Mar 15 09:36 rtl8723befw_36.bin
-rw-r--r-- 1 root root 20886 Mar 15 09:36 rtl8723bs_ap_wowlan.bin
-rw-r--r-- 1 root root  9120 Mar 15 09:36 rtl8723bs_bt.bin
-rw-r--r-- 1 root root 32108 Mar 15 09:36 rtl8723bs_nic.bin
-rw-r--r-- 1 root root 26398 Mar 15 09:36 rtl8723bs_wowlan.bin
-rw-r--r-- 1 root root 20886 Mar 15 09:36 rtl8723bu_ap_wowlan.bin
-rw-r--r-- 1 root root 32108 Mar 15 09:36 rtl8723bu_nic.bin
-rw-r--r-- 1 root root 26398 Mar 15 09:36 rtl8723bu_wowlan.bin
-rw-r--r-- 1 root root 11662 Mar 15 09:36 rtl8723fw.bin
-rw-r--r-- 1 root root 22996 Mar 15 09:36 rtl8723fw_B.bin
-rw-r--r-- 1 root root 28984 Mar 15 09:36 rtl8821aefw.bin
-rw-r--r-- 1 root root 28348 Mar 15 09:36 rtl8821aefw_29.bin
-rw-r--r-- 1 root root 19858 Mar 15 09:36 rtl8821aefw_wowlan.bin
```

```
root@c2b874e:/usr/src/app# lspci -knn
00:00.0 Host bridge [0600]: Intel Corporation Atom Processor Z36xxx/Z37xxx Series SoC Transaction Register [8086:0f00] (rev 0f)
        Subsystem: Device [1027:2014]
        Kernel driver in use: iosf_mbi_pci
00:02.0 VGA compatible controller [0300]: Intel Corporation Atom Processor Z36xxx/Z37xxx Series Graphics & Display [8086:0f31] (rev0f)
        Subsystem: Device [1027:2014]
        Kernel driver in use: i915
00:1a.0 Encryption controller [1080]: Intel Corporation Atom Processor Z36xxx/Z37xxx Series Trusted Execution Engine [8086:0f18] (rev 0f)
        Subsystem: Intel Corporation Atom Processor Z36xxx/Z37xxx Series Trusted Execution Engine [8086:0f18]
00:1d.0 USB controller [0c03]: Intel Corporation Atom Processor Z36xxx/Z37xxx Series USB EHCI [8086:0f34] (rev 0f)
        Subsystem: Intel Corporation Atom Processor Z36xxx/Z37xxx Series USB EHCI [8086:0f34]
        Kernel driver in use: ehci-pci
00:1f.0 ISA bridge [0601]: Intel Corporation Atom Processor Z36xxx/Z37xxx Series Power Control Unit [8086:0f1c] (rev 0f)
        Subsystem: Intel Corporation Atom Processor Z36xxx/Z37xxx Series Power Control Unit [8086:0f1c]
```

floion, 2018-04-22:
Hi,
can you also share the boot log please?

robp2175, 2018-04-23:
```
23.04.18 11:53:03 (-0400) Installing service 'main sha256:fc2ced6cb01b82779b875c0176ad1912f0a7d705b56f0aea9259252a356d47b7'
23.04.18 11:53:04 (-0400) Installed service 'main sha256:fc2ced6cb01b82779b875c0176ad1912f0a7d705b56f0aea9259252a356d47b7'
23.04.18 11:53:04 (-0400) Starting service 'main sha256:fc2ced6cb01b82779b875c0176ad1912f0a7d705b56f0aea9259252a356d47b7'
23.04.18 11:53:04 (-0400) Started service 'main sha256:fc2ced6cb01b82779b875c0176ad1912f0a7d705b56f0aea9259252a356d47b7'
23.04.18 11:53:04 (-0400) /sbin/udevd
23.04.18 11:53:04 (-0400) + URL=https://www.emcc.edu
23.04.18 11:53:04 (-0400) + echo 10.40.1.69 concerto.emcc.edu
23.04.18 11:53:04 (-0400) + echo 10.40.1.49 dashboard.emcc.edu
23.04.18 11:53:04 (-0400) + umount /dev/shm
23.04.18 11:53:04 (-0400) + python -m SimpleHTTPServer 8888
23.04.18 11:53:05 (-0400) + mount -t tmpfs shm /dev/shm
23.04.18 11:53:05 (-0400) + export PASS=
23.04.18 11:53:05 (-0400) + PASS=
23.04.18 11:53:05 (-0400) + chpasswd
23.04.18 11:53:05 (-0400) + echo root:
23.04.18 11:53:05 (-0400) ++ openssl passwd -1 -stdin
23.04.18 11:53:05 (-0400) ++ echo kiosk
23.04.18 11:53:05 (-0400) + useradd -u 12345 -g users -d /home/kiosk -s /bin/bash -p '$1$v54Ig4AT$Ffh0ysOephMVC8xV/S1vD.' kiosk
23.04.18 11:53:05 (-0400) + usermod -aG sudo kiosk
23.04.18 11:53:05 (-0400) + export PASS=
23.04.18 11:53:05 (-0400) + PASS=
23.04.18 11:53:05 (-0400) + echo kiosk:
23.04.18 11:53:05 (-0400) + chpasswd
23.04.18 11:53:05 (-0400) Serving HTTP on 0.0.0.0 port 8888 ...
23.04.18 11:53:05 (-0400) + useradd -m pi
23.04.18 11:53:05 (-0400) + gpasswd -a pi video
23.04.18 11:53:05 (-0400) Adding user pi to group video
23.04.18 11:53:06 (-0400) + rm -rf /tmp/resin
23.04.18 11:53:06 (-0400) rm: cannot remove ‘/tmp/resin’: Device or resource busy
23.04.18 11:53:06 (-0400) + bash /usr/src/app/cups.sh
23.04.18 11:53:06 (-0400) cupsctl: Unable to connect to server: Bad file descriptor
23.04.18 11:53:06 (-0400) + service cron start
23.04.18 11:53:06 (-0400) [....] Starting periodic command scheduler: cron[ ok .
23.04.18 11:53:06 (-0400) + bash /usr/src/app/job.sh
23.04.18 11:53:06 (-0400) giblib error: Can't open X display. It *is* running, yeah?
23.04.18 11:53:06 (-0400) Screenshot Taken
23.04.18 11:53:06 (-0400) + '[' '' == Kiosk ']'
23.04.18 11:53:06 (-0400) + rm /tmp/.X0-lock
23.04.18 11:53:06 (-0400) + true
23.04.18 11:53:06 (-0400) + pid=146
23.04.18 11:53:06 (-0400) + wait 146
23.04.18 11:53:06 (-0400) + startx /usr/src/app/node_modules/electron/dist/electron /usr/src/app/electron-browser --enable-logging
23.04.18 11:53:06 (-0400) hostname: No address associated with hostname
23.04.18 11:53:06 (-0400) xauth:  file /root/.Xauthority does not exist
23.04.18 11:53:06 (-0400) xauth: (stdin):1:  bad display name "c2b874e:0" in "add" command
23.04.18 11:53:06 (-0400) 
23.04.18 11:53:06 (-0400) /etc/X11/xinit/xserverrc: 3: exec: /usr/bin/X: not found
23.04.18 11:53:21 (-0400) xinit: giving up
23.04.18 11:53:21 (-0400) xinit: unable to connect to X server: Connection refused
23.04.18 11:53:21 (-0400) xinit: server error
23.04.18 11:53:21 (-0400) xauth: (argv):1:  bad display name "c2b874e:0" in "remove" command
23.04.18 11:53:21 (-0400) Service exited 'main sha256:fc2ced6cb01b82779b875c0176ad1912f0a7d705b56f0aea9259252a356d47b7'
23.04.18 11:53:22 (-0400) Restarting service 'main sha256:fc2ced6cb01b82779b875c0176ad1912f0a7d705b56f0aea9259252a356d47b7'
23.04.18 11:53:04 (-0400) /sbin/udevd
23.04.18 11:53:04 (-0400) + URL=https://www.emcc.edu
23.04.18 11:53:04 (-0400) + echo 10.40.1.69 concerto.emcc.edu
23.04.18 11:53:04 (-0400) + echo 10.40.1.49 dashboard.emcc.edu
23.04.18 11:53:04 (-0400) + umount /dev/shm
23.04.18 11:53:04 (-0400) + python -m SimpleHTTPServer 8888
23.04.18 11:53:05 (-0400) + mount -t tmpfs shm /dev/shm
23.04.18 11:53:05 (-0400) + export PASS=
23.04.18 11:53:05 (-0400) + PASS=
23.04.18 11:53:05 (-0400) + chpasswd
23.04.18 11:53:05 (-0400) + echo root:
23.04.18 11:53:05 (-0400) ++ openssl passwd -1 -stdin
23.04.18 11:53:05 (-0400) ++ echo kiosk
23.04.18 11:53:05 (-0400) + useradd -u 12345 -g users -d /home/kiosk -s /bin/bash -p '$1$v54Ig4AT$Ffh0ysOephMVC8xV/S1vD.' kiosk
23.04.18 11:53:05 (-0400) + usermod -aG sudo kiosk
23.04.18 11:53:05 (-0400) + export PASS=
[... 192 lines cut ...]
23.04.18 11:53:39 (-0400) + PASS=
23.04.18 11:53:39 (-0400) + echo kiosk:
23.04.18 11:53:39 (-0400) + chpasswd
23.04.18 11:53:39 (-0400) + useradd -m pi
23.04.18 11:53:39 (-0400) useradd: user 'pi' already exists
23.04.18 11:53:39 (-0400) + rm -rf /tmp/resin
23.04.18 11:53:39 (-0400) rm: cannot remove ‘/tmp/resin’: Device or resource busy
23.04.18 11:53:39 (-0400) + bash /usr/src/app/cups.sh
23.04.18 11:53:39 (-0400) cupsctl: Unable to connect to server: Bad file descriptor
23.04.18 11:53:39 (-0400) mkdir: cannot create directory ‘/data/cifs’: File exists
23.04.18 11:53:40 (-0400) + service cron start
23.04.18 11:53:40 (-0400) [....] Starting periodic command scheduler: cron[ ok .
23.04.18 11:53:40 (-0400) + bash /usr/src/app/job.sh
23.04.18 11:53:40 (-0400) giblib error: Can't open X display. It *is* running, yeah?
23.04.18 11:53:40 (-0400) Screenshot Taken
23.04.18 11:53:40 (-0400) + '[' '' == Kiosk ']'
23.04.18 11:53:40 (-0400) + rm /tmp/.X0-lock
23.04.18 11:53:40 (-0400) + true
23.04.18 11:53:40 (-0400) + pid=126
23.04.18 11:53:40 (-0400) + wait 126
23.04.18 11:53:40 (-0400) + startx /usr/src/app/node_modules/electron/dist/electron /usr/src/app/electron-browser --enable-logging
23.04.18 11:53:40 (-0400) hostname: No address associated with hostname
23.04.18 11:53:40 (-0400) xauth: (stdin):1:  bad display name "c2b874e:0" in "add" command
23.04.18 11:53:40 (-0400) 
23.04.18 11:53:40 (-0400) /etc/X11/xinit/xserverrc: 3: exec: /usr/bin/X: not found

floion, 2018-05-03:
Hi,
I meant the output f when running "dmesg" in the hostOS webterminal

robp2175, 2018-05-03:
Ha, sorry, that certainly makes more sense. I have updated the response above

robp2175, 2018-05-03:
I can also grant you access if you want.

alexgg, 2021-08-06:
Hi, `lspci` should be identifying the hardware. If it cannot be seen in the PCI bus adding the driver is not going to do much.

If this is still a problem a suggest you bring the issue through our forums support in https://forums.balena.io where our team will help you out.

closed 2021-08-06 by alexgg