balena-os/balena-intel #143: Fix booting from resin-image

agherzan, 2018-08-13:
The resin-image is currently unbootable. This image is not bootable as UEFI image nor as non-UEFI one. The resin-image used to be bootable in the past (can't remember what was the default mode).

floion, 2018-08-30:
@agherzan what would you like the bootloader on the resin-image to be? grub EFI or grub legacy?

agherzan, 2018-08-30:
I would expect it to be the default one we have for the flasher

floion, 2018-08-30:
@agherzan the flasher has them both. So you want resin-image to have them both?

agherzan, 2018-08-30:
I miss context sadly here. In the past we used to have this working. Why did we remove it from the normal image? Again, I'd expect the resin-image to boot as the flasher one does. So preferably both modes supported.

floion, 2018-08-30:
We had some boards that cannot boot with both options enabled

agherzan, 2018-08-30:
I see. In that case I'd vote for having them both by default and let the flasher remove one or another.

floion, 2018-08-31:
@agherzan https://github.com/resin-os/resin-intel/pull/151
Can you give it a test?

agherzan, 2018-09-05:
@floion I've pushed a fix and it works well now. Thanks.