balena-os/balena-intel #163: Getting CPU Temperature in /sys

rgaufman, 2018-10-25:
Hi there,

I am trying your Resin img file on an Intel NUC and while it is working beautifully, I am not able to find the relevant files in /sys to check the CPU temperature.

I had a look and it seems these files are missing in /lib/modules:

```
/lib/modules/*/kernel/drivers/hwmon/coretemp.ko
/lib/modules/*/kernel/drivers/hwmon/nct6775.ko
```

Could these please be added in the next version?

floion, 2018-12-03:
@shaunmulligan again another request which will increase the rootfs size.
Any progress on meta-balena about being able to increase the rootfs size in future releases?

rgaufman, 2019-01-15:
Any update on this? 🙂

floion, 2019-01-18:
Hi, no update yet. We created https://github.com/balena-os/meta-balena/issues/1366 so when we have the rootfs smaller only then we can include additional drivers in.

rgaufman, 2019-03-11:
It seems the other ticket was done, can this be added now?

floion, 2019-03-11:
Hi. Thanks for the nudge. Will PR it so that it will be in 2.31.0+rev2 or newer

floion, 2019-03-11:
This should be fixed by https://github.com/balena-os/balena-intel/pull/193

closed 2019-03-14 by floion