balena-os/balena-intel #169: Update getting started link

shaunmulligan, 2018-11-14:
https://github.com/balena-os/resin-intel/blob/f0df498b5324a946941ea50ebacc993f300745a2/genericx86-64.coffee#L38

should be changed to point to https://www.balena.io/docs/learn/getting-started/intel-nuc/nodejs/

floion, 2019-02-01:
@shaunmulligan so shall we change the getting started links for all the boards in the same fashion as you suggested here?

floion, 2019-12-14:
@garethtdavies are you the person to go about this sort of bugs?

garethtdavies, 2019-12-16:
@floion yes! I've opened a PR for this and I'll go through the other repos. Specifically, on this one there are no Surface getting started docs so, for now, are the NUC docs the best place?

floion, 2020-01-31:
I think so @garethtdavies so let's leave it like that for the moment.
Done in https://github.com/balena-os/balena-intel/pull/282

closed 2020-01-31 by floion