balena-os/balena-intel #177: some firmware not present in the /lib/firmware after build image

ashishka, 2018-11-30:
Hello!

I am have builded image for genericx86-64 but some needed firmware not present here, but it need to use wifi on new inter brix. Here is this firmware:

iwlwifi-3168-21.ucode
iwlwifi-3168-22.ucode
iwlwifi-3168-27.ucode
iwlwifi-3168-29.ucode

I am added this manually, but i sure, many other people need it to be added to the
./layers/meta-resin/meta-resin-sumo/recipes-core/packagegroups/packagegroup-resin-connectivity.bbappend
./layers/poky/meta/recipes-kernel/linux-firmware/linux-firmware_git.bb 
inside this repo https://github.com/balena-os/resin-intel

Thank you!

floion, 2018-11-30:
Hi. Thanks for the heads-up. We'll need to do some checking on the size this adds. We are already running low in the rootfs partition.

ashishka, 2018-11-30:
This is will increase rootfs in the ~4MB

floion, 2019-02-01:
Hi. For what wifi chipset is this for?

ashishka, 2019-02-06:
Hi!
We use this wifi for connect gigabyte brix to lan and internet.

floion, 2019-12-14:
4 MB is too much for us to include at this point. balena OS 3.x will either have a bigger rootfs or allow for some mechanism to let individual boards declare a need for bigger rootfs sizes. So until then, we are closing this issue

closed 2019-12-14 by floion