balena-os/balena-intel #182: Firmware missing for Intel Dual Band Wireless-9560

thundron, 2019-01-03:
The Intel Dual Band Wireless-AC 9560 firmware was supposed to be included in v2.27, but it looks like we did not do that, as we did not properly manage the flow of the previously reported issue.
The needed firmware is `iwlwifi-9000-pu-b0-jf-b0-34.ucode`

pdcastro, 2019-01-18:
I understand that work on this ticket is blocked by another ticket:
https://github.com/balena-os/meta-balena/issues/1366

That other ticket has the purpose of freeing up space in the relevant system partition so that more firmwares can be shipped for the NUC with the currently released OS versions (I understand that balenaOS 2.29 is also affected). balenaOS 3.0 will attack the problem from a different angle: increasing the partition size.

pdcastro, 2019-01-18:
Forum thread reporting the problem on 2.29.0+rev1:
https://forums.balena.io/t/intel-nuc-wifi-doesnt-work/5041/6

floion, 2019-01-31:
Fixed by https://github.com/balena-os/balena-intel/pull/187

floion, 2019-02-01:
Closing issue as the firmware will be available in release v2.29.2+rev2 or newer

closed 2019-02-01 by floion