balena-os/balena-intel #202: Re-flashing requires erasing existing partitions manually

kazazes, 2019-04-18:
This behavior is new. I have had to erase partition tables in between flashes or it stalls. Reproduced on two different model NUCs.

floion, 2019-04-19:
What error do you get? Does it complain about overlapping partitions?

kazazes, 2019-04-19:
Yes. In the flash sequence and in GParted afterwards.

floion, 2019-04-29:
@agherzan @ZubairLK seems like some trouble with the labeling ^

agherzan, 2019-04-29:
@spanceac had a similar issue that he fixed and might be able to give us more details. Otherwise, @kazazes do you have an easy way to reproduce?

floion, 2019-04-29:
@agherzan the way @spanceac fixed it a while back was to wipe the partition table from u-boot shell (it happened on an armv7 board running uboot and the u-boot shell was available). Also, @acostach fixed another instance of this bug happening with this PR: https://github.com/balena-os/meta-balena/pull/1379

So it looks like there are other cases when this can happen

floion, 2019-05-01:
Related to this: https://github.com/balena-os/meta-balena/pull/1487

floion, 2019-05-16:
Should be fixed by https://github.com/balena-os/meta-balena/pull/1506 and https://github.com/balena-os/balena-intel/pull/210

closed 2019-05-21 by floion

kazazes, 2019-08-24:
I can confirm that this is still in fact still present. Trying to flash a `2.38.3+rev4` dev image to a production `2.39.0+rev3` hangs at `Booting flash` with no more information. Wiping manually with another live image fixes the problem.

agherzan, 2019-09-09:
@kazazes are you able to reproduce this consistently? Can you give us some instructions to replicate it on our side? CC @floion