balena-os/balena-intel #203: Missing firmware for built-in Intel NUC Bluetooth

pdcastro, 2019-04-24:
Placeholder issue for further investigation:

_Intel NUC built in bluetooth controller not recognized_
https://forums.balena.io/t/intel-nuc-built-in-bluetooth-controller-not-recognized/6441
```
[ 3.603081] bluetooth hci0: Direct firmware load for intel/ibt-18-16-1.sfi failed with error -2
[ 3.603086] Bluetooth: hci0: Failed to load Intel firmware file (-2)
```

pdcastro, 2019-04-25:
Attached: full dmesg output
[dmesg-NUC-BLKNUC7i7DNH1E.txt](https://github.com/balena-os/balena-intel/files/3116366/dmesg-NUC-BLKNUC7i7DNH1E.txt)

resin-ci, 2019-04-26:
[pdcastro] This issue has attached support thread https://jel.ly.fish/#/support-thread~f822d4e5-1336-4901-8e1d-96487d34f0b4

floion, 2019-04-29:
Fixed by https://github.com/balena-os/balena-intel/pull/206

closed 2019-04-29 by floion