balena-os/balena-intel #259: shutdown command is delayed

floion, 2019-10-17:
In latest OS releases (2.44.0+rev1) the shutdown command takes a while to complete.

krixkrix, 2019-11-07:
I had the same observation.
I talked to a Balena supporter who pointed me to this, so I will repeat my observations:

Rebooting my devices has started to take looooong time. Around 2 minutes from the reboot command until the device is seen again. It seems to be the shutdown phase that is the bottleneck. 

The problem is observer when I press "Reboot" in the Balena dashboard. It is NOT observed when just pressing "Restart". It is also observed when commanding a shutdown through the supoervisor API.

Here is a Balena log snippet showing the 2 minute delay:
```
07.11.19 12:10:41 (+0100) Killing service 'main sha256:309970af6fc68d72532a9d10a7a6647530430b9f8217b714d4c5ce44f9ab1f34'
07.11.19 12:10:47 (+0100) Killed service 'main sha256:309970af6fc68d72532a9d10a7a6647530430b9f8217b714d4c5ce44f9ab1f34'
07.11.19 12:10:47 (+0100) Rebooting
07.11.19 12:10:47 (+0100) Service exited 'main sha256:309970af6fc68d72532a9d10a7a6647530430b9f8217b714d4c5ce44f9ab1f34'
07.11.19 12:12:53 (+0100) Supervisor starting
```

ZubairLK, 2019-11-11:
@krixkrix traced it to a systemd service upstream. More detail here
https://git.yoctoproject.org/cgit/cgit.cgi/poky/commit/meta/recipes-support/rng-tools?h=warrior&id=95632e679df13fd5a45725e3752c49986ed2fe73

internal ref
https://www.flowdock.com/app/rulemotion/resin-devices/threads/rI2F9FgFxfjjfv2IK7wuTfUfvyI

bash one-liner via hostOS
**exercise caution**
```
mount -o remount,rw / && \
sed -i 's/Before=sysinit.target/Before=sysinit.target shutdown.target\nConflicts=shutdown.target/g' /lib/systemd/system/rngd.service /lib/systemd/system/rngd.service.new && \
mv /lib/systemd/system/rngd.service.new /lib/systemd/system/rngd.service && \ 
systemctl daemon-reload && \
mount -o remount,ro /
```

floion, 2019-11-13:
Fixed by https://github.com/balena-os/balena-intel/pull/261

closed 2019-11-13 by floion