balena-os/balena-intel #283: Validate on Intel NUC 10th gen

Ereski, 2020-01-30:
The NUC image should work on a 10th gen, but apparently it doesn't.

JF: https://jel.ly.fish/fb396cce-c42f-4e71-94c7-c60192d70e98

shaunmulligan, 2020-01-31:
Hardware sourced from https://www.cdw.com/product/intel-next-unit-of-computing-kit-10-performance-nuc10i7fnh-mini-pc-co/5881287

floion, 2020-05-18:
For this Intel NUC variant one needs to disable Modern Standby. Just enter BIOS and set sleep type to Legacy S3 standby. It is available under POWER -> Secondary Power Settings -> Sleep Type support.
Wifi will work on balena OS release newer than 2.50.0 but eth will need to wait for a release with kernel >= 5.5

asininemonkey, 2020-10-17:
Any timeframe on a kernel 5.5+ release? Another owner of an Intel NUC 10 here with no ethernet (wireless not an option for me).

chriswiggins, 2020-11-01:
Hi, we've just hit this issue with a large amount of Intel NUC 10th gens we have - can we please have an ETA on 5.5?

chriswiggins, 2020-12-09:
Is there a way we can see what the progress is behind this Jellyfish support?

klvnptr, 2020-12-10:
We also need ethernet support for 10th gen Intel NUC. Could you please give us a timeline? Thank you.

floion, 2021-02-05:
Hi all, the latest OS release `2.68.1+rev1` should have support for this. Can you please confirm this on your side?

asininemonkey, 2021-02-05:
Just upgraded my NUC and switched back to the internal ethernet port and it appears to be working. Will monitor closely but so far so good. Many thanks. 🙂

closed 2021-02-12 by floion

adamtec, 2021-07-09:
I have two Intel NUC 10th gen as well as two Intel NUC 8th gen that I've put on the same Balena app. I've noticed that the temperature reported by Balena is around 80C for the 10th gen while 8th gen is around 50C. Both devices are hardly working with about 2% CPU usage and similarly equipped (storage and ram). As a result, the fans are very noisy on the 10th gen devices. 8th gen devices are NUC8i7BNH and 10th gen devices are NUC10i7FNH. I can provide additional details if useful.