balena-os/balena-intel #300: bump kernel version or backport ovl regression fix

robertgzr, 2020-05-06:
setting `OVERLAY_FS_INDEX=n` or using the overlay module parameter `index=off` does not sufficiently relax the checks for overlapping layers, which breaks the overlay2 graphdriver in docker

see https://github.com/balena-io/balena/issues/2105

we're using http://git.yoctoproject.org/cgit/cgit.cgi/linux-yocto/log/?h=v5.2/standard/base&id=8d8bf56b2373bfdb1a9702b96a0e91e6706f62d4

but we need the patch here http://git.yoctoproject.org/cgit/cgit.cgi/linux-yocto/log/?h=v5.2/standard/base&id=d7aa8c546ab97b9011f805042b3647693fce107f

robertgzr, 2020-05-06:
@floion this is essentially the same as https://github.com/balena-os/balena-raspberrypi/pull/462

floion, 2020-05-13:
Fixed by https://github.com/balena-os/balena-intel/pull/302

closed 2020-05-29 by robertgzr