balena-os/balena-intel #320: Realtek Audio Drivers not present

harborretail, 2020-07-22:
I'm looking for support for the chtrt5645 (Intel CherryTrail audio chipset)

Can this module be added into the intel-image? Or do I need to build this kernel module on my own?

floion, 2020-08-17:
Hi. I don't think we can currently add new modules to the Intel NUC image.
We do however have an extended image in which we could add them, then Generic x86-64 ext device type. Can you use this device type instead?

harborretail, 2020-08-17:
@floion I have been in contact with Jordan and he directed me to that image as well.

Yes, I have been using this image (I didn't realize you had the genericx86-64 device type with the larger kernel).

We can close this out thank you!

closed 2020-08-17 by harborretail