balena-os/balena-intel #334: Release nvidia GPU drivers as a hostapp extension

alexgg, 2020-10-23:
Customers with Nvidia GPUs are having trouble building the drivers as external modules. We should prepare a hostapp not only with the drivers but also the required libraries to use the GPU for CUDA etc.

This depends on the work on hostapp extension support to be completed

openedhardware, 2020-10-26:
Just wanted to keep this issue tracked.

Any idea how long it would take to fix this? :+1:

alexgg, 2020-10-26:
hi @openedhardware we are currently working on this same support for the Nvidia Jetson device types. Once that is done we will start working on this.

openedhardware, 2021-06-03:
Hi, @alexgg 
Any update here?

alexgg, 2021-06-08:
hi @openedhardware unfortunately the scope of this work keeps growing and we don't have a clear ETA at the moment.

maxekman, 2023-06-13:
I would too like to know if any work has gone into this lately? Would be great to have!