balena-os/balena-intel #340: Intel NUC production image won't boot, but development does

klvnptr, 2020-12-12:
I downloaded the latest production "balenaOS 2.50.1+rev1" image, and installed in on a new Intel NUC 10 NUC10I5FNH2.

I get the exact same error message, that sighmon reported in this forum post:
https://forums.balena.io/t/dell-precision-3640-not-getting-past-the-balena-logo/202120
I also uploaded a screenshot here: https://tinyimg.io/i/EaWl6rb.jpg

I downloaded the same image, but development version. In this case everything works. (except the ethernet port, but we are using USB-Ethernet dongle as a workaround).

Can you guys please give me some pointers, what could be wrong? 

Thank you

alexgg, 2021-08-06:
Could you please retry with a recent version? The Intel NUC 10 should already be supported.

klvnptr, 2021-08-06:
we have put another OS on our NUCs for now, but when we install new ones we will try it. btw. i have read in other comments that it works with 10th gen NUCs, so i happily close this issue.

closed 2021-08-06 by klvnptr