balena-os/balena-intel #464: Wifi is not working for Intel NUC 11 (NUC11ATKC4)

jaomaloy, 2022-04-12:
An Intel NUC 11 (NUC11ATKC4) wifi is not working. These messages are in the dmesg logs:
[ 2.231554] iwlwifi 0000:00:14.3: no suitable firmware found!
[ 2.231567] iwlwifi 0000:00:14.3: minimum version required: iwlwifi-QuZ-a0-jf-b0-39
[ 2.231569] iwlwifi 0000:00:14.3: maximum version supported: iwlwifi-QuZ-a0-jf-b0-59
[ 2.231571] iwlwifi 0000:00:14.3: check git://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git

dfunckt, 2022-04-12:
WiFi seems to be working on NUC 11 with genericx86-64-ext.

mtoman, 2022-04-13:
We have merged the support for wifi6 adapters recently but apparently some 11th gen NUCs with a wifi5 Intel Wireless-AC 9462.

alexgg, 2022-07-04:
This support was added in v2.98.15.