balena-os/balena-intel #590: i915 firmware missing for Kaby Lake (KBL) GPUs

jakogut, 2022-12-05:
User reports that GPU accelerated video decode does not work, as the i915 driver is unable to load GuC firmware.

```
[ 5.180037] i915 0000:00:02.0: Direct firmware load for i915/kbl_guc_62.0.0.bin failed with error -2
[ 5.180041] i915 0000:00:02.0: [drm] GuC firmware i915/kbl_guc_62.0.0.bin: fetch failed with error -2
[ 5.180043] i915 0000:00:02.0: [drm] GuC firmware(s) can be downloaded from https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/i915
[ 5.180666] i915 0000:00:02.0: [drm] GuC is uninitialized
```

Fixed by: https://github.com/balena-os/meta-balena/pull/2897

cakoolen, 2023-01-11:
Is there any progress on this story? I can see that the PR is slowly getting old but no real comments are made to it.

closed 2023-02-24 by alexgg

cakoolen, 2023-02-25:
I don't see how this problem is fixed now since I cannot update my devices to a newer OS version let alone one that actually fixes the issue.

commit 1f1b1e12 (2023-02-24): linux-firmware: Add newer firmware for Kaby Lake GPUs

The proper fix would be to use upstream firmware in meta-balena but
that keeps getting delayed and there are people actually blocked by this.
This commit can be rolled back after the meta-balena PR is sorted out.

Fixes: #590

Change-type: patch
Changelog-entry: Add newer firmware for Kaby Lake GPUs