balena-os/balena-intel #633: Requests for kernel >= 5.16 for Alder Lake GPUs

mtoman, 2023-05-03:
i915 does not officially support Alder Lake GPUs until kernel 5.16.

Older kernels can load the driver with a parameter (e.g. `modprobe i915 force_probe=4680`) for partial functionality, which in practice seems to be good enough but needs the user application to remove the autoloaded driver and then load it back with the parameter.

klvnptr, 2023-06-26:
hello Guys. any update on this?

klvnptr, 2023-06-27:
@alexgg can you please let me know when you guys upgrade kernel to 5.16 for alder lake gpu support? thank you.

alexgg, 2023-06-29:
hey @klvnptr there is a dichotomy between the `intel NUC` and the `Generic x86_64` device types. For modern devices, Intel NUC included, we would recommend using the  `Generic x86_64 (GPT)` device type as it's the one that is receiving most attention, including secure boot support. We have an ongoing task to update that device type to a 6.x kernel.

We will be discussing internally how to move customers away from the `Intel NUC` device type and into the `Generic x86_64` ones to avoid the team having to maintain both. We would also need a migration path defined for current customers, so it's not likely to happen soon, but that is the current thinking.

klvnptr, 2023-06-29:
hey @alexgg kindly thank you for your reply. secure boot is something that is very long awaited. can't wait to see that as well in release.

in the meantime could you please point us how to rebuild the generic x86 64 image with 5.16 kernel? building the image seem to be straightforward (thanks for the awesome docs), but need some pointers for bumping kernel from 5.15 to 5.16. thank you.

alexgg, 2023-06-29:
hey @klvnptr glad to know you are interested in secure boot and disk encryption - the feature is already available in production, check https://blog.balena.io/balenaOS-secure-boot-and-disk-encryption-for-x86-64/

Updating from v5.15 to v5.16 I would expect it to be similar to https://github.com/balena-os/balena-generic/commit/2e376f07447a9199315bfe463987a1240b0397b8.

However, we always try to update to LTS (Long Term Support) kernel releases, and v5.16 is not one of them (v5.15 and v6.1 are).

klvnptr, 2023-07-03:
thanks @alexgg excellent news. do you have an estimated timeline for 6.1 kernel? thanks.

alexgg, 2023-07-03:
hey @klvnptr it's being worked on in https://github.com/balena-os/balena-generic/pull/244 but there is no timeline I'm afraid.

spinda, 2024-12-30:
@alexgg It looks like that was resolved some time ago. Any idea why Intel NUC OS images are still shipping with a 5.15 kernel?

alexgg, 2024-12-30:
hey @spinda, with Intel having discontinued the Intel NUC, there is no appetite to update it. Our intention is to discontinue it in the short term and to migrate existing customer to using the `Generic x86_64 (GPT)` images on NUC hardware. Those have an updated kernel and new features like secure boot support.

We are just pending on having a no fuss migration path ready to discontinue the Intel NUC.