balena-os/balena-intel #645: Request for Kernel 6.1

rmorillo24, 2023-05-22:
The request comes in order to have iGPU support for AMD Ryzen 9 7900

hgaiser, 2023-05-23:
I'm curious, why is this issue in the `balena-intel` repository and not the `balena-generic` repository?

rmorillo24, 2023-05-23:
@hgaiser we are tracking in this repo the requests related to intel issues. It may affect in the future a different repo, but at the moment this is the origin. I changed the title to make it clearer, and in line with previous similar requests

hgaiser, 2023-06-13:
Is there any update on this issue? A beta / release-candidate would also work for us.

alexgg, 2023-07-07:
Relates-to:
* https://github.com/balena-os/balena-generic/pull/244
* https://github.com/balena-os/balena-generic/pull/260
* https://github.com/balena-os/balena-generic/pull/263

alexgg, 2023-07-07:
hey @hgaiser please keep an eye on the PRs above - once merged a new release will be deployed for the `Generic x86_64` device types and you can use those to test with.

hgaiser, 2023-08-01:
@alexgg it seems the PRs have been merged (or superseded by other PRs). When can we expect a new build for `Generic x86_64`? It seems that balenaOS 3.1.4 still uses 5.15.54.

alexgg, 2023-08-01:
hey @hgaiser, v3.1.4 of `Generic x86_64 (GPT)` has a 6.1 kernel - are you referring to the legacy MBR device type?

hgaiser, 2023-08-01:
Ah that's strange. I didn't install it, but I downloaded https://files.balena-cloud.com/images/genericx86-64-ext/3.1.4/kernel_modules_headers.tar.gz which seems to point to 5.15.54. Are these not the correct kernel module headers for `Generic x86_64 (GPT)`?

alexgg, 2023-08-01:
hey @hgaiser the `genericx86-64-ext` is the machine name for the `Generic x86_64 (legacy MBR)` device type. You want the `generic-amd64` machine name. 
https://files.balena-cloud.com/images/generic-amd64/3.1.4/kernel_modules_headers.tar.gz

hgaiser, 2023-08-02:
Thanks @alexgg, that seems to be the one :).