balena-os/balena-intel #759: Request to enable PINCTRL_BROXTON support in the kernel for Apollo Lake processor

memoryrs, 2024-08-14:
We are running balenaOS on a Motium TUFF Panel PC which recently upgraded to Apollo Lake processor.
The Apollo Lake GPIO watchdog driver depends on PINCTRL_BROXTON; hence we need this config to be enabled in order to use watchdog.
I've created a pull request #758 (https://github.com/balena-os/balena-intel/pull/758) that implements this.

alexgg, 2024-08-14:
Hi, thanks for bringing this to our attention.

Did you check whether the `Generic x86_64 (GPT)` device type enables this configuration? We are advising customers that start projects to use that device type instead. The `Intel NUC` has been deprecated by Intel and although it is still a supported platform is it maintained for MBR/BIOS devices. UEFI/GPT devices should use the Generic x86_64 (GPT) images instead.

memoryrs, 2024-08-14:
Yes, the Generic x86_64 image is what I built on. If you have a look at my pull request https://github.com/balena-os/balena-intel/pull/758/files - that is what I am modifying?

alexgg, 2024-08-14:
hey @memoryrs it's confusing, there is a `Generic x86_64 (GPT)` device type (machine name `generic-amd64`) in https://github.com/balena-os/balena-generic/blob/master/generic-amd64.coffee which is where our generic x86_64 support is.

The `Intel NUC` device type (machine name genericx86-64) is what you are currently using, which is fine, except that we have discussed discontinuing that specific device type in line with Intel's HW discontinuation. So if you are starting a new product, it may be better to start with the  `Generic x86_64 (GPT)`  device.

In any case we can review and merge this change, it's just a side note.

memoryrs, 2024-08-19:
Thank you @alexgg for clarification, and thanks for reviewing this change. At the moment, we wish to continue using the `Intel NUC` device type (machine name genericx86-64). So we'd really appreciate getting this merged to allow us to continue the work.

We'd also like to have the ability to migrate to the `Generic x86_64 (GPT)` device in the future due to the discontinuation you mentioned. Hence we'll need the same change for https://github.com/balena-os/balena-generic/blob/master/generic-amd64.coffee. I'll request a new issue along with a PR on https://github.com/balena-os/balena-generic if that's okay?

alexgg, 2024-08-19:
@memoryrs The `Generic x86_64 (GPT)` image already configures `CONFIG_PINCTRL_BROXTON=m`, that's why I mentioned it initially.

floion, 2024-08-19:
For the `Intel NUC` device type I created https://github.com/balena-os/balena-intel/pull/762 from your PR @memoryrs

memoryrs, 2024-08-20:
Thanks heaps for the useful info @alexgg. Would you kindly point me out the location of these configurations? I was looking at https://github.com/balena-os/balena-intel/blob/master/layers/meta-balena-genericx86/recipes-kernel/linux/genericx86-64-ext/defconfig for the `Intel NUC` and https://github.com/balena-os/balena-generic/blob/master/layers/meta-balena-generic/recipes-kernel/linux/files/generic-amd64/defconfig for the `Generic x86_64 (GPT)` at first; but realised they were obsolete auto generated files. My apologies I'm still a newbie on the Yocto Project.

Thank you for creating https://github.com/balena-os/balena-intel/pull/762 for this change @floion

floion, 2024-08-28:
Fixed by https://github.com/balena-os/balena-intel/pull/762

closed 2024-08-28 by floion